nadavmatalon/MCP3221 #1: Other processors

jcullins0, 2017-11-14:
This is really not an issue per sei but, It would be great for the lib to work with other processors like the STM32F103  and the ESP32.
I am currently designing a project that has to be isolated to protect the processor from high voltage noise.
I want to use this chip and an I2C isolator from TI. I am using the STM32F103.
The functions you have added are just what I need.
Thank you for the great work.
JC

nadavmatalon, 2017-11-20:
Hi JC,

Sorry for the delay in reply and thanks for the kind words! Very happy you found this lib useful!

To your comment, making the lib compatible with other processors would indeed be useful, if only I could find the time to get it done :-)

That said, I've added it to my (very long) to-do list and would hopefully get to it at some point in the future (though regretfully I can't say when exactly).

All the best,
Nadav

closed 2017-11-20 by nadavmatalon

jcullins0, 2017-11-20:
Thank you so much,
I know you are very busy and have your priorities, so whenever you get the
time would be great.
JC