nadavmatalon/MCP3221 #4: Lib Compatible with Esp8266?

laercionit, 2019-03-11:
The Lib is Compatible with Esp8266?

nadavmatalon, 2019-03-12:
I don't know, never tried.

laercionit, 2019-03-12:
I am getting compile error because LIB is for AVR architecture only, I believe a little help to solve it, since it is all by I2C using Wire.

laercionit, 2019-03-12:
Change architectures=avr to architectures=*

Remove :
#if !defined(ARDUINO_ARCH_AVR)
#error “The MCP3221 library only supports AVR processors.”
#endif

afagard, 2019-10-29:
> Change architectures=avr to architectures=*
> 
> Remove :
> #if !defined(ARDUINO_ARCH_AVR)
> #error “The MCP3221 library only supports AVR processors.”
> #endif

and with that change, does it work with your esp?

nadavmatalon, 2024-02-07:
> Yes it does, I can confirm that after removing from `MCP3221.h`
> 
> ```
> #if !defined(ARDUINO_ARCH_AVR)
> #error “The MCP3221 library only supports AVR processors.”
> #endif
> ```
> 
> The library compiles and works fine on ESP8266-12. Not sure why author blocked it only for AVR processors.

I did it because when I wrote the library ages ago there were other architectures I couldn't test with and didn't want to risk it. Back then, the ESP's weren't available/common yet so were not taken into consideration. 

Glad it works though! If you come across this issue, just remove the restriction as suggested above (I'll remove it from the library permenantly when I have the time :-)

closed 2024-02-07 by nadavmatalon