nadavmatalon/MCP3221 #5: question regarding vref

afagard, 2019-10-31:
You have this line `(_vRef / (float)DEFAULT_VREF)` in `getVoltage()` and I was just wondering what the idea was regarding this. It is my understanding that `_vRef` defaults to `4096` as does DEFAULT_VREF. So this would then be 1. However, say you have a 5v reference, then this becomes 5000/4096 = 1.22? Why? I didn't see that the MCP had some internal set state at 4096 so I'm just wondering why this calculation is performed.

nadavmatalon, 2019-11-01:
Hi afagard,

Happy you found the lib useful.

Had a quick look at the datasheet of the MCP3221, and although I've written the lib a long time ago and it's hard to remember, I do believe the reason for the value of 4096 for the vRef does stem from a factory-set value in the chip itself (so if you use 5V reference and want to keep the measurements precise an adjustment needs to be made in the calculation).

Have a look at the formula in section 4.1 of the datasheet (page 14) and hopefully it will make sense. The datasheet can be found here: http://ww1.microchip.com/downloads/en/devicedoc/20001732e.pdf

Hope this helps.

afagard, 2019-11-02:
Yup. Missed that one! Thanks

closed 2019-11-02 by afagard