nadavmatalon/MCP3221 #6: Multiple MCP3221 in same sketch

yapluka128, 2020-12-16:
Hello,

I'm trying to use 2 MCP in the same sketch, but meet no success. Both have different addresses and are working well alone (both connected to the same bus, but sketch is talking to only one at a time).
So wiring is good, I already tried to switch to brand new design.
here is my simple test code :
`#include "MCP3221.h"
#include "utility/MCP3221InfoStr.h"

const byte DEV_ADDR_1 = 0x4C;                       // I2C address of the MCP3221 (Change as needed)
const byte DEV_ADDR_2 = 0x4D;                       // I2C address of the MCP3221 (Change as needed)

MCP3221 mcp3221_1(DEV_ADDR_1);
MCP3221 mcp3221_2(DEV_ADDR_2);

void setup() {
    Serial.begin(115200);
    Wire.begin();
    while(!Serial);
    Serial.print(F("\n\nCURRENT SETTINGS:\n"));
    Serial.print(MCP3221InfoStr(mcp3221_1));
    Serial.print(F("\nCHANGING TO NEW SETTINGS..."));
    mcp3221_1.setVref(5112);
    mcp3221_1.setSmoothing(ROLLING_AVG);
    mcp3221_1.setRes1(10251);
    mcp3221_1.setRes2(4705);
    mcp3221_1.setVinput(VOLTAGE_INPUT_12V);
    mcp3221_1.setAlpha(134);
    mcp3221_1.setNumSamples(16);
    Serial.print(F("DONE\n"));
    Serial.print(MCP3221InfoStr(mcp3221_1));
    Serial.print(F("\nRESETTING DEVICE..."));
    mcp3221_1.reset();
    Serial.print(F("DONE\n"));
    Serial.print(MCP3221InfoStr(mcp3221_1));
    Serial.print(F("\n\n"));
    delay(1000);
    Serial.println("Change");
    Serial.print(F("\n\nCURRENT SETTINGS:\n"));
    Serial.print(MCP3221InfoStr(mcp3221_2));
    Serial.print(F("\nCHANGING TO NEW SETTINGS..."));
    mcp3221_2.setVref(5112);
    mcp3221_2.setSmoothing(ROLLING_AVG);
    mcp3221_2.setRes1(10251);
    mcp3221_2.setRes2(4705);
    mcp3221_2.setVinput(VOLTAGE_INPUT_12V);
    mcp3221_2.setAlpha(134);
    mcp3221_2.setNumSamples(16);
    Serial.print(F("DONE\n"));
    Serial.print(MCP3221InfoStr(mcp3221_2));
    Serial.print(F("\nRESETTING DEVICE..."));
    mcp3221_2.reset();
    Serial.print(F("DONE\n"));
    Serial.print(MCP3221InfoStr(mcp3221_2));
    Serial.print(F("\n\n"));
}`

And the result :
`15:45:58.931 -> MCP3221 DEVICE INFORMATION
15:45:58.931 -> --------------
15:45:58.931 -> 
15:45:58.931 -> CURRENT SETTINGS:
15:45:58.931 -> 
15:45:58.931 -> MCP3221 DEVICE INFORMATION
15:45:58.931 -> --------------------------
15:45:58.931 -> I2C ADDRESS:	   0 (0)
15:45:58.931 -> I2C COM STATUS:	   NOT CONNECTED
15:45:58.931 -> I2C ERROR:	 Error Code #2: Address sent, NACK received
15:45:58.931 -> 
15:45:58.931 -> CHANGING TO NEW SETTINGS...DONE
15:45:58.931 -> 
15:45:58.931 -> MCP3221 DEVICE INFORMATION
15:45:58.931 -> --------------------------
15:45:58.931 -> I2C ADDRESS:	   0 (0)
15:45:58.931 -> I2C COM STATUS:	   NOT CONNECTED
15:45:58.977 -> I2C ERROR:	 Error Code #2: Address sent, NACK received
15:45:58.977 -> 
15:45:58.977 -> RESETTING DEVICE...DONE
15:45:58.977 -> 
15:45:58.977 -> MCP3221 DEVICE INFORMATION
15:45:58.977 -> --------------------------
15:45:58.977 -> I2C ADDRESS:	   0 (0)
15:45:58.977 -> I2C COM STATUS:	   NOT CONNECTED
15:45:58.977 -> I2C ERROR:	 Error Code #2: Address sent, NACK received
15:45:58.977 -> 
15:45:58.977 -> 
15:45:59.954 -> Change
15:45:59.954 -> 
15:45:59.954 -> 
15:45:59.954 -> CURRENT SETTINGS:
15:46:00.000 -> 
15:46:00.000 -> MCP3221 DEVICE INFORMATION
15:46:00.000 -> --------------------------
15:46:00.000 -> I2C ADDRESS:	   77 (0X4D)
15:46:00.000 -> I2C COM STATUS:	   CONNECTED
15:46:00.000 -> VOLTAGE REFERENCE: 4096mV
15:46:00.000 -> SMOOTHING METHOD:  EMAVG
15:46:00.000 -> VOLTAGE INPUT:	   5V
15:46:00.000 -> VD RESISTOR 1:	   0R
15:46:00.000 -> VD RESISTOR 2:	   0R
15:46:00.000 -> ALPHA:		   178
15:46:00.000 -> SAMPLES BUFFER:	   10 SAMPLES
15:46:00.000 -> 
15:46:00.000 -> CHANGING TO NEW SETTINGS...DONE
15:46:00.000 -> 
15:46:00.000 -> MCP3221 DEVICE INFORMATION
15:46:00.000 -> --------------------------
15:46:00.000 -> I2C ADDRESS:	   77 (0X4D)
15:46:00.000 -> I2C COM STATUS:	   CONNECTED
15:46:00.000 -> VOLTAGE REFERENCE: 5112mV
15:46:00.000 -> SMOOTHING METHOD:  ROLLING-AVAREGE
15:46:00.000 -> VOLTAGE INPUT:	   12V
15:46:00.000 -> VD RESISTOR 1:	   10251R
15:46:00.000 -> VD RESISTOR 2:	   4705R
15:46:00.000 -> ALPHA:		   134
15:46:00.000 -> SAMPLES BUFFER:	   16 SAMPLES
15:46:00.047 -> 
15:46:00.047 -> RESETTING DEVICE...DONE
15:46:00.047 -> 
15:46:00.047 -> MCP3221 DEVICE INFORMATION
15:46:00.047 -> --------------------------
15:46:00.047 -> I2C ADDRESS:	   77 (0X4D)
15:46:00.047 -> I2C COM STATUS:	   CONNECTED
15:46:00.047 -> VOLTAGE REFERENCE: 4096mV
15:46:00.047 -> SMOOTHING METHOD:  EMAVG
15:46:00.047 -> VOLTAGE INPUT:	   5V
15:46:00.047 -> VD RESISTOR 1:	   0R
15:46:00.047 -> VD RESISTOR 2:	   0R
15:46:00.047 -> ALPHA:		   178
15:46:00.047 -> SAMPLES BUFFER:	   10 SAMPLES
15:46:00.047 -> 
15:46:00.047 -> `

It looks that the first instantiated device is overwritten by the second one. 
What did I miss? 
I'm using the last version of the library.

nadavmatalon, 2020-12-17:
Hi,

I'm sorry, but I've written this library so long ago I can hardly remember anything about it. 
One thing that caught my eye in the example sketch you included are these lines:
mcp3221_1.reset();
mcp3221_2.reset();
Why are you reseting the MCP3221s to default values after setting all the parameters? Could this be the problem?

nikowot, 2024-09-26:
Hello, nadavmatalon
I report the same trouble as ‘yapluka128’

I don't use the command ‘mcp3221_1.reset();’ in my code, but I can't handle the second and each further instance correctly.

I have several MCP3221 chips on my bus and each one handled individually - works fine.

Example for the first MCP3221 chip with address [0x48]:

The I2c bus analysis shows the transmission sequence:

Read [0x48] + ACK
[01]+ACK
[A3]+NAK
The resulting [01A3]hex value is 419dec, or 0.419V. This is the true value, confirmed by voltmeter measurement.

When I add several instances of MCP3221, the communication falls down :-(

In the I2c bus analysis with two MCP3221 chips, the transmission sequence is shown:

(Setup) Read [0x4A] + ACK
[00]+ACK
[00]+NAK

(Setup) Read [0x4A] + ACK
[00]+ACK
[00]+NAK

I expected a sequence like this:
Read [0x48] + ACK
[01]+ACK
[A3]+NAK

Read [0x4A] + ACK
[01]+ACK
[A3]+NAK

Please analyse , where is the problem?