nadavmatalon/MCP3221 #7: MCP3221 voltage reading floating with basic usage example

yeong78, 2024-02-07:
Hi, I use MCP3221 ADC to read my voltage sensor differential output .
But when i run it with MCP3221 basic usage example, i found out that the result floating around.
Example the input for Ain =1.7Vrms (sin wave) but the result float around 1.7V to 1V.
Anyone have any idea why my read out result is floating around?
Is it because my Ain input is Sin wave or cause by noise? VDD input =5V.
[MCP3221 ADC design.xlsx](https://github.com/nadavmatalon/MCP3221/files/14190983/MCP3221.ADC.design.xlsx)

nadavmatalon, 2024-02-07:
Hi, sorry to hear you're having trouble with this. I have no idea what could be the issue as I haven't encountered this kind of behaviour. The only suggestion I can make is to find a different MCP3221 library and try it out to see if the problem recurs. If it does, it may indicate a hardware problem (e.g. a bad connection) rather than a problem with a specific library. Hope this helps.

yeong78, 2024-02-16:
Hi, Now i direct supply the Ain with dc supply and found that the result have some tolerance (0.6V) with the actual Ain value.
Here i attached my schematic and multiple Ain input with the output reading for your reference.
 MY design for the Ain input is without voltage divider resistor, i direct supply the Ain with dc supply.

[image]